CH3NGYZ/tailscale-openwrt #5: tailscale 占用体积太大

hotwa, 2023-09-17:
tailscale_1.48.2_mips.tgz 下载22M解压40多M，一般用的闪存芯片都是32M的。我用的newifi3-d2。
固件编译有1.44，但是有些版本不一致有些功能无法使用。怎么解决？是否可以小一点

CH3NGYZ, 2023-09-17:
虽然安装包是这么大, 但是脚本是直接解压到/tmp里面的, 相当于解压到内存里面,  你的内存大于64M就可以存的下的

hotwa, 2023-09-20:
路由器会定时重启怎么办？那安装的tailscale不就没了。还有openwrt的tailscale 1.44版本我用的时候出口节点无法生效，防火墙什么的都开了，应该不是打洞的问题。我在windows和mac下设置出口节点都很正常

CH3NGYZ, 2023-09-20:
> 路由器会定时重启怎么办？那安装的tailscale不就没了。还有openwrt的tailscale 1.44版本我用的时候出口节点无法生效，防火墙什么的都开了，应该不是打洞的问题。我在windows和mac下设置出口节点都很正常

不会呀, 路由器每次重启会自动下载最新版本的文件, 配置还在的

hotwa, 2023-09-20:
有gitee的仓库吗？需要魔法

CH3NGYZ, 2023-09-20:
> 路由器会定时重启怎么办？那安装的tailscale不就没了。还有openwrt的tailscale 1.44版本我用的时候出口节点无法生效，防火墙什么的都开了，应该不是打洞的问题。我在windows和mac下设置出口节点都很正常

install脚本会下载一个相当于一个下载器的脚本放在/usr/bin/tailscale, 然后使它开机自启,  /usr/bin/tailscale作用是/tmp/tailscale不存在则下载最新版

CH3NGYZ, 2023-09-20:
> 有gitee的仓库吗？需要魔法

https://github.com/CH3NGYZ/tailscale-openwrt/blob/chinese_mainland/README.md

hotwa, 2023-09-20:
tailscale和clash冲突吗

CH3NGYZ, 2023-09-20:
> tailscale和clash冲突吗

有可能会冲突, 最好的方法是修改clash规则,  DOMAIN-SUFFIX,tailscale.com 为直连

hotwa, 2023-09-20:
` wget --tries=5 -c -t 60 -O- https://raw.githubusercontent.com/CH3NGYZ/tailscale-openwrt/main/install.sh | sh
--2023-09-20 11:24:06--  https://raw.githubusercontent.com/CH3NGYZ/tailscale-openwrt/main/install.sh
Resolving raw.githubusercontent.com... 185.199.111.133, 185.199.109.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2482 (2.4K) [text/plain]
Saving to: 'STDOUT'

-                            100%[===========================================>]   2.42K  --.-KB/s    in 0.001s  

2023-09-20 11:24:08 (3.89 MB/s) - written to stdout [2482/2482]

The current architecture of the machine is mipsle|, which scripts are compatible with, so leave a comment on this issue so that the author can modify the architecture parts of the documentation: https://github.com/CH3NGYZ/tailscale-openwrt/issues/6
wait for 5 seconds
Unknown package 'libustream-openssl'.
Package ca-bundle (20230311-1) installed in root is up to date.
Package kmod-tun (5.15.127-1) installed in root is up to date.
Collected errors:
 * opkg_install_cmd: Cannot install package libustream-openssl.
--2023-09-20 11:24:14--  https://ghproxy.com/https://raw.githubusercontent.com/CH3NGYZ/tailscale-openwrt/chinese_mainland/tailscale-openwrt.tgz
Resolving ghproxy.com... 192.9.132.155
Connecting to ghproxy.com|192.9.132.155|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1426 (1.4K) [application/octet-stream]
Saving to: 'tailscale-openwrt.tgz'

tailscale-openwrt.tgz        100%[===========================================>]   1.39K  --.-KB/s    in 0s      

2023-09-20 11:24:17 (31.8 MB/s) - 'tailscale-openwrt.tgz' saved [1426/1426]

usr/
usr/bin/
usr/bin/tailscale
usr/bin/tailscaled
etc/
etc/init.d/
etc/init.d/tailscale
/etc/rc.d/K1tailscale   /etc/rc.d/S80tailscale  /etc/rc.d/S99tailscale
Please wait, Tailscaled service is downloading the Tailscale executable file in the background......
The download timed out. You need to manually open the system log to view the failure cause...`
哪里看日志？

CH3NGYZ, 2023-09-20:
系统日志

> ` wget --tries=5 -c -t 60 -O- https://raw.githubusercontent.com/CH3NGYZ/tailscale-openwrt/main/install.sh | sh --2023-09-20 11:24:06-- [...]

```logread | grep tailscale```

CH3NGYZ, 2023-09-20:
> ` wget --tries=5 -c -t 60 -O- https://raw.githubusercontent.com/CH3NGYZ/tailscale-openwrt/main/install.sh | sh --2023-09-20 11:24:06-- [...]

 看你这个报错是因为没有检测到安装文件, 你手动执行以下命令: usr/bin/tailscale  看看什么结果

CH3NGYZ, 2023-09-20:
> ` wget --tries=5 -c -t 60 -O- https://raw.githubusercontent.com/CH3NGYZ/tailscale-openwrt/main/install.sh | sh --2023-09-20 11:24:06-- [...]

我知道什么原因了, 因为我的仓库中没有mipsle架构的文件, 导致无法下载, 稍等我添加完毕你再重新运行

hotwa, 2023-09-20:
```shell
`[root@OpenWrt:03:09 PM ~] # wget --tries=5 -c -t 60 -O- https://ghproxy.com/https://raw.githubusercontent.com/CH3NGYZ/tailscale-openwrt/chinese_mainland/install.sh | sh
--2023-09-20 15:10:11--  https://ghproxy.com/https://raw.githubusercontent.com/CH3NGYZ/tailscale-openwrt/chinese_mainland/install.sh
Resolving ghproxy.com... 150.136.252.237
Connecting to ghproxy.com|150.136.252.237|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2252 (2.2K) [text/plain]
Saving to: 'STDOUT'

-                            100%[===========================================>]   2.20K  --.-KB/s    in 0s      

2023-09-20 15:10:14 (43.7 MB/s) - written to stdout [2252/2252]

当前机器的架构是mipsle|mips, 脚本兼容此架构, 请在这个issue留下评论以便作者及时修改说明文档中支持的架构部分: https://github.com/CH3NGYZ/tailscale-openwrt/issues/6
等待5秒
Unknown package 'libustream-openssl'.
Package ca-bundle (20230311-1) installed in root is up to date.
Unknown package 'kmod-tun'.
Collected errors:
 * opkg_install_cmd: Cannot install package libustream-openssl.
 * opkg_install_cmd: Cannot install package kmod-tun.
--2023-09-20 15:10:20--  https://ghproxy.com/https://raw.githubusercontent.com/CH3NGYZ/tailscale-openwrt/chinese_mainland/tailscale-openwrt.tgz
Resolving ghproxy.com... 192.9.132.155
Connecting to ghproxy.com|192.9.132.155|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1426 (1.4K) [application/octet-stream]
Saving to: 'tailscale-openwrt.tgz'

tailscale-openwrt.tgz        100%[===========================================>]   1.39K  --.-KB/s    in 0s      

2023-09-20 15:10:23 (32.1 MB/s) - 'tailscale-openwrt.tgz' saved [1426/1426]

usr/
usr/bin/
usr/bin/tailscale
usr/bin/tailscaled
etc/
etc/init.d/
etc/init.d/tailscale
/etc/rc.d/K1tailscale   /etc/rc.d/S99tailscale
请等待, Tailscaled 服务正在后台下载 Tailscale 可执行文件...
/tmp/tailscaled 存在, 继续
如果无法登陆, 请运行 '/etc/init.d/tailscale stop && clear && /usr/bin/tailscaled' 命令检查日志, 或重新运行 tailscale up
failed to connect to local tailscaled; it doesn't appear to be running
failed to connect to local tailscaled; it doesn't appear to be running`
```

hotwa, 2023-09-20:
```shell
[root@OpenWrt:03:10 PM ~] # tailscale --version
1.48.2
  tailscale commit: a6bcfd69149c491d7542cc758b762bae8882db04
  other commit: 62c580b04f6f7f04ae58c71e1096e51b6f18e6c6
  go version: go1.21.0
[root@OpenWrt:03:12 PM ~] # tailscale up
failed to connect to local tailscaled; it doesn't appear to be running
```

CH3NGYZ, 2023-09-20:
> ```shell
> /etc/init.d/tailscale stop && clear && /usr/bin/tailscaled
> ```

/etc/init.d/tailscale stop && clear && /usr/bin/tailscaled 看看什么情况

CH3NGYZ, 2023-09-20:
> ```shell
> [root@OpenWrt:03:10 PM ~] # tailscale --version
> 1.48.2
>   tailscale commit: a6bcfd69149c491d7542cc758b762bae8882db04
>   other commit: 62c580b04f6f7f04ae58c71e1096e51b6f18e6c6
>   go version: [...]

是不是还有原来的旧版本残留影响

CH3NGYZ, 2023-09-20:
> ```shell
> [root@OpenWrt:03:10 PM ~] # tailscale --version
> 1.48.2
>   tailscale commit: a6bcfd69149c491d7542cc758b762bae8882db04
>   other commit: 62c580b04f6f7f04ae58c71e1096e51b6f18e6c6
>   go version: [...]

```
logread | grep tailscaled
```
运行这行, 贴一下结果

hotwa, 2023-09-20:
```shell
[root@OpenWrt:08:36 AM /tmp/root] # logread | grep tailscaled
Failed to connect to ubus
```

CH3NGYZ, 2023-09-20:
> ```shell
> 
> [root@OpenWrt:08:36 AM /tmp/root] # logread | grep tailscaled
> 
> Failed to connect to ubus
> 
> ```

luci界面的系统日志看一下有没有tailscaled的日志，看看什么原因
现在是已经安装完毕了，至于打不开就是其他的问题了，你得想办法google一下

hotwa, 2023-09-25:
重装 openwrt 系统后已解决，可以整理一下卸载脚本的内容。重复安装提示卸载不干净，我重装解决的

hotwa, 2023-09-25:
如果可以加一个 openwrt 的derper服务器也是可行的

CH3NGYZ, 2023-09-25:
> 如果可以加一个 openwrt 的derper服务器也是可行的

卸载脚本提示卸载不干净能截张图吗, 还有derper可以直接修改tailscale配置的吧? 我没有自建所以不太了解

CH3NGYZ, 2023-09-25:
> 重装 openwrt 系统后已解决，可以整理一下卸载脚本的内容。重复安装提示卸载不干净，我重装解决的

卸载是不是有很多/sys/device的设备? 例如device tailscale0

hotwa, 2023-10-03:
我直接重新将系统还原重置了，不过建议还是搞一个卸载的 shell 脚本。现在已经成功了，这个issue 可以关闭了

hotwa, 2023-10-03:
登录命令行参考：
`tailscale up --login-server=http://your.domin.top:58080 --accept-routes=true --accept-dns=false --advertise-routes=192.168.11.0/24 --advertise-exit-node --netfilter-mode=off`

`--netfilter-mode=off`默认关闭防火墙,如下：

`echo 'net.ipv4.ip_forward = 1' | tee /etc/sysctl.d/ipforwarding.conf
echo 'net.ipv6.conf.all.forwarding = 1' | tee -a /etc/sysctl.d/ipforwarding.conf
sysctl -p /etc/sysctl.d/ipforwarding.conf`

需要开启转发`ipv4`

防火墙配置参考：
`uci add network tailscale
uci set network.tailscale=interface
uci set network.tailscale.proto=none
uci set network.tailscale.device=tailscale0
uci commit network

uci add firewall zone
uci set firewall.@zone[-1]=zone
uci set firewall.@zone[-1].name=tailscale
uci set firewall.@zone[-1].input=ACCEPT
uci set firewall.@zone[-1].output=ACCEPT
uci set firewall.@zone[-1].forward=ACCEPT
uci set firewall.@zone[-1].masq=1
uci set firewall.@zone[-1].mtu_fix=1
uci set firewall.@zone[-1].network=tailscale
uci commit firewall

uci add firewall forwarding
uci set firewall.@forwarding[-1].src=tailscale
uci set firewall.@forwarding[-1].dest=wan
uci commit firewall

uci add firewall forwarding
uci set firewall.@forwarding[-1].src=lan
uci set firewall.@forwarding[-1].dest=tailscale
uci commit firewall

uci add firewall forwarding
uci set firewall.@forwarding[-1].src=tailscale
uci set firewall.@forwarding[-1].dest=lan
uci commit firewall`

检查：
网络--->接口，添加新接口，
名称：tailscale
协议：不配置协议
设备：tailscale0

hotwa, 2023-10-03:
shell创建 `tailscale0`网卡似乎是会失败，建议手动添加：

>root@OpenWrt:11:35 AM ~] # /etc/init.d/tailscale stop && clear && /usr/bin/tailscaled
logtail started
Program starting: v1.50.1-tf45c02bfc-g36a20760a, Go 1.21.1: []string{"/tmp/tailscaled"}
LogID: 00901a07698bc03c65567a02d11d32491a88bd1963a52a2cd66d42b044bc438a
logpolicy: using system state directory "/var/lib/tailscale"
wgengine.NewUserspaceEngine(tun "tailscale0") ...
Linux kernel version: 5.15.132
is CONFIG_TUN enabled in your kernel? `modprobe tun` failed with: 
Missing required package kmod-tun; run: opkg install kmod-tun
wgengine.NewUserspaceEngine(tun "tailscale0") error: tstun.New("tailscale0"): CreateTUN("tailscale0") failed; /dev/net/tun does not exist
flushing log.
logger closing down
logtail: upload: log upload of 923 bytes compressed failed: Post "https://log.tailscale.io/c/tailnode.log.tailscale.io/101fb1483b192173a2750a077fa76dc02d8c20e008a776d0e88bad9c18969c40": context canceled
getLocalBackend error: createEngine: tstun.New("tailscale0"): CreateTUN("tailscale0") failed; /dev/net/tun does not exist

hotwa, 2023-10-03:
[image]

CH3NGYZ, 2023-10-03:
好的感谢，我现在在老家，等到了公司再处理

CH3NGYZ, 2023-10-03:
> shell创建 `tailscale0`网卡似乎是会失败，建议手动添加：

似乎是因为没有kmod-tun这个软件包？试试opkg install kmod-tun是不是抱错无法安装？因为脚本内有自动安装过，应该是当时抱错了，但不明显，你可能没注意到，看你日志的这里：
Linux kernel version: 5.15.132
is CONFIG_TUN enabled in your kernel? modprobe tun failed with:
Missing required package kmod-tun; run: opkg install kmod-tun

closed 2023-10-17 by CH3NGYZ