CH3NGYZ/tailscale-openwrt #6: Does the script support your machine architecture? 脚本是否支持您的机器架构?

CH3NGYZ, 2023-09-19:
Leave a comment containing the output of `uname -m` and whether it was installed and running successfully.
留下评论，包含 'uname -m' 的输出，以及Tailscale是否安装成功、运行成功。

hotwa, 2023-09-25:
mipsle|mips installed successfully and running successfully.

the tailscaled save in /tmp.

monkeyotg, 2023-10-16:
x86_64|amd64 installed successfully and running successfully

CH3NGYZ, 2023-10-17:
> x86_64|amd64 installed successfully and running successfully

收到, 感谢您的回复

AKAEL0817, 2024-03-06:
arch_:aarch64| arch:arm64 安装成功，运行成功

wangweitung, 2024-03-27:
INSTALL: ---------------------------------------------
当前机器的架构是 arch_:x86_64| arch:amd64

AKAEL0817, 2024-03-27:
支持

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

cero-ss, 2024-04-13:
ARMv8|arm 安装成功 运行成功

CH3NGYZ, 2024-05-22:
> i686 安装失败

请附上详细安装日志, 我也不知道i686对应的文件名叫什么

CH3NGYZ, 2024-08-20:
closed, 项目趋于稳定, 可以不用再提交了

closed 2024-08-20 by CH3NGYZ