CH3NGYZ/tailscale-openwrt #7: 代理连不上呀

gzxiexl, 2023-11-11:
[image]

CH3NGYZ, 2023-11-11:
可能我自建的代理用不了， 稍等修改成另外一个

CH3NGYZ, 2023-11-11:
已修复

closed 2023-11-11 by CH3NGYZ

gzxiexl, 2023-11-11:
echo "请等待,超时时间为三分钟, Tailscaled 服务正在后台下载 Tailscale 可执行文件..."

到这一步就挂了😭

CH3NGYZ, 2023-11-11:
> echo "请等待,超时时间为三分钟, Tailscaled 服务正在后台下载 Tailscale 可执行文件..."
> 
> 
> 到这一步就挂了😭

可能用的还是之前的cloudflare地址，看看/usr/bin/tailscale文件内是不是还是workers.dev那个代理，如果是的话你手动对比我的内容修改一下，因为存在缓存的关系，所以你下载的脚本可能不是最新的。

gzxiexl, 2023-11-12:
[image][image]

还是下载不动的情况

gzxiexl, 2023-11-12:
> > echo "请等待,超时时间为三分钟, Tailscaled 服务正在后台下载 Tailscale 可执行文件..."
> > 
> > 
> > 到这一步就挂了😭
> 
> 手动运行 `tailscale` 报什么错呢

[image]
手动下载的网速太慢了🥲

gzxiexl, 2023-11-12:
https://www.jsdelivr.com/github

能否套个这个

CH3NGYZ, 2023-11-12:
> https://www.jsdelivr.com/github
> 
> 能否套个这个

正在写

CH3NGYZ, 2023-11-12:
> https://www.jsdelivr.com/github
> 
> 能否套个这个

测试，换了另外的proxy， jsdrlivr在这里用不了

gzxiexl, 2023-11-12:
终于能用了✌️

CH3NGYZ, 2023-11-13:
> 终于能用了✌️

也不知道现在这个代理能用多久, 失效了再联系我替换~

CH3NGYZ, 2023-12-23:
增加了代理池 (虽然不多), 同时写了"下载超时切换代理"以及"下载失败切换代理"的功能, 现在应该会大大减少下载慢以及下载失败无法启动tailscale的问题了, 请使用最新版本安装脚本重新安装一遍.
[image]

24kbrother, 2024-07-28:
出口节点不能用！>> /etc/resolv.conf  

之前的脚本都没问题，最近更新了一次吧，一周左右，新安装到openwrt的tailscale，会自动修改/etc/resolv.conf 内容如下：
root@OpenWrt:~# cat /etc/resolv.conf 
search lan
nameserver 100.100.100.100
nameserver 223.5.5.5
nameserver 119.29.29.29

这里的DNS服务器多了100.100.100.100会导致使用出口节点的设备无法正常访问Internet，所有网页都打不开，起初以为是防火墙问题，但当我排查到 /etc/resolv.conf文件的时候找到了原因，请UP修复一下，急需啊！

这里可以手动删除'nameserver 100.100.100.100'但是一旦openwrt重启就又会自动更新！！