CH3NGYZ/tailscale-openwrt #8: reboot后版本未升级 保留在1.48.2了

gzxiexl, 2023-11-30:


gzxiexl, 2023-11-30:
usr/bin/tailscale
里面的代码 版本号还是1.48.2 是不是这个问题

CH3NGYZ, 2023-11-30:
> usr/bin/tailscale
> 
> 里面的代码 版本号还是1.48.2 是不是这个问题

是的，代码中写死如果获取不到最新版本，则使用内置的版本号，我待会先检查一下这个代理是不是失效了，代理失效，404获取不到版本就是默认的1.48.2（因为这个版本下的release中有所有架构的包，不会因为找不到这个架构的安装包而安装失败）

CH3NGYZ, 2023-11-30:
我实测这个cdn是可以使用的, 或许是其他原因导致获取不到版本号? 可以多重启几次看看, 接下来我也会把默认版本从 1.48.2改为最新的1.54.0

CH3NGYZ, 2023-11-30:
[image]
2023.11.30 16:25:00 UTC+8测试通过

closed 2023-11-30 by CH3NGYZ

CH3NGYZ, 2023-12-01:
> 我可能是因为代理地址访问不了 reboot 后 从 1.54 倒退回 1.48 了 如果要手动设计怎么操作？

或者修改/速溶/bin/tailscale和tailscaled中的1.48.1到1.54.1

gzxiexl, 2023-12-18:
> > 我可能是因为代理地址访问不了 reboot 后 从 1.54 倒退回 1.48 了 如果要手动设计怎么操作？
> 
> 卸载，使用最新版本的安装脚本安装，最新版的版本号设置成了1.54.0

晕了，安装最新1.56 reboot后退回到1.54了
系统日志我截取一段：
```
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  24500K .......... .......... .......... .......... .......... 97% 18.4M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  24550K .......... .......... .......... .......... .......... 97% 13.1M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  24600K .......... .......... .......... .......... .......... 97% 12.7M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  24650K .......... .......... .......... .......... .......... 97% 6.63M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  24700K .......... .......... .......... .......... .......... 98% 16.1M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  24750K .......... .......... .......... .......... .......... 98% 5.80M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  24800K .......... .......... .......... .......... .......... 98% 18.8M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  24850K .......... .......... .......... .......... .......... 98% 13.2M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  24900K .......... .......... .......... .......... .......... 98% 13.7M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  24950K .......... .......... .......... .......... .......... 99% 13.9M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  25000K .......... .......... .......... .......... .......... 99% 17.3M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  25050K .......... .......... .......... .......... .......... 99% 10.9M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  25100K .......... .......... .......... .......... .......... 99% 11.9M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  25150K .......... .......... .......... .......... .......... 99% 11.6M 0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:  25200K .......... .......... ...                             100% 3.29M=4.0s
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]: 2023-12-18 10:53:04 (6.19 MB/s) - written to stdout [25829053/25829053]
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]:
Mon Dec 18 10:53:04 2023 daemon.info tailscaled[8209]: 下载完毕!
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]: 2023/12/18 02:53:04 logtail started
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]: 2023/12/18 02:53:04 Program starting: v1.54.0-t42cae6916-g2458acf00, Go 1.21.4: []string{"/tmp/tailscaled", "--port", "41641", "--state", "/etc/config/tailscaled.state", "--statedir", "/etc/tailscale/"}
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]: 2023/12/18 02:53:04 LogID: a0bc3de9eac6cefbae4c5e149a5fae65b5504fa371e859ce6afdaafad98a681e
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]: 2023/12/18 02:53:04 logpolicy: using system state directory "/var/lib/tailscale"
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]: logpolicy.ConfigFromFile /var/lib/tailscale/tailscaled.log.conf: open /var/lib/tailscale/tailscaled.log.conf: no such file or directory
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]: logpolicy.Config.Validate for /var/lib/tailscale/tailscaled.log.conf: config is nil
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]: 2023/12/18 02:53:04 wgengine.NewUserspaceEngine(tun "tailscale0") ...
Mon Dec 18 10:53:04 2023 daemon.err tailscaled[8209]: 2023/12/18 02:53:04 router: default choosing iptables

```

gzxiexl, 2023-12-18:
能否支持直接更新，更新成功后再reboot

CH3NGYZ, 2023-12-18:
> 能否支持直接更新，更新成功后再reboot

我下一步完善一下自动切换代理, 这个联不通换下一个, 你现在可以直接修改本地的脚本 /usr/bin/tailscale和tailscaled,把里面的版本号改一下

CH3NGYZ, 2023-12-23:
现在可以了, 写了 下载超时切换代理 以及 下载失败切换代理 功能, 现在应该会大大减少获取不到最新版本号的问题了
[image]