CH3NGYZ/tailscale-openwrt #9: 一直获取不到版本号，是代理的问题么

iMaxxxxxxx, 2024-01-20:
[image]

iMaxxxxxxx, 2024-01-20:
[image]

CH3NGYZ, 2024-01-20:
为啥看不见安装日志呢，奇怪
你是不是用的passwall，clash等翻墙工具

CH3NGYZ, 2024-01-20:
> [image]
> 

为啥看不见安装日志呢，好奇怪

iMaxxxxxxx, 2024-01-20:
昨天还运行了一次代理安装，有安装日志，然后最后一步启动前台程序的时候卡住了
我之前装的是一个1.26还是多少版本的，也是通过你的脚本安装的
是有PassWall和ShadowSocksR Plus+这两个是启用状态
[image]

CH3NGYZ, 2024-01-20:
> 昨天还运行了一次代理安装，有安装日志，然后最后一步启动前台程序的时候卡住了
> 我之前装的是一个1.26还是多少版本的，也是通过你的脚本安装的
> 是有PassWall和ShadowSocksR Plus+这两个是启用状态
> [image]
> 
> 

我觉得可能是代理的问题，关掉代理，卸载重启后重新安装

iMaxxxxxxx, 2024-01-20:
好的，我试试。
我运行了rm -rf /tmp/tailscale是不是旧版本一起删掉了，昨天执行卸载命令，也是没有日志

iMaxxxxxxx, 2024-01-20:
[image]

CH3NGYZ, 2024-01-20:
> 好的，我试试。
> 我运行了rm -rf /tmp/tailscale是不是旧版本一起删掉了，昨天执行卸载命令，也是没有日志
> 

那你把安装脚本下下来之后，手动给予执行权限，然后用sh 脚本.sh执行看看什么日志

iMaxxxxxxx, 2024-01-20:
关掉代理重启后运行的
[image]

CH3NGYZ, 2024-01-20:
> 关掉代理重启后运行的
> [image]
> 

你把后面的 竖杠 sh删掉，只留wget链接，删掉选项里的q，只留下大写O，然后看下载脚本的日志

iMaxxxxxxx, 2024-01-20:
[image]

CH3NGYZ, 2024-01-20:
> [image]
> 

那就是脚本下不下来咯，试试curl www.baidu.com什么效果，如果还连不通，那你网络环境有问题，如果能联通，那就是ghproxy不好使

CH3NGYZ, 2024-01-20:
> [image]
> 

感觉你的dna有问题，检查下吧

iMaxxxxxxx, 2024-01-20:
确实是我路由表有问题，添加之后可以正常卸载并重新安装新版本了，谢谢大佬

CH3NGYZ, 2024-01-20:
> 确实是我路由表有问题，添加之后可以正常卸载并重新安装新版本了，谢谢大佬
> 

ok的没问题，issue关掉了

closed 2024-01-20 by CH3NGYZ

iMaxxxxxxx, 2024-01-20:
最后有个报错
[image]

CH3NGYZ, 2024-01-20:
> 最后有个报错
> [image]
> 

你是不是同一个账号下其他设备开了路由模式？那这样是正常的

iMaxxxxxxx, 2024-01-21:
感谢感谢