CH3NGYZ/tailscale-openwrt #10: 尊敬的作者，关于istoreOS更新就出错的问题

WEEWXX, 2024-02-16:
经过测试，ARM架构软路由，用贵方提供的脚本，新装，卸载，重启均无问题；但是一但tailscale有更新，路由器重启就会导致路由器自身无法上网，提示dns错误，子网设备一切正常，但是检查配置没有找到哪儿出错，当故障出现，卸载、重装均故障依旧。烦请提供一下处理故障思路。

CH3NGYZ, 2024-02-17:
请检查是不是翻墙工具导致路由表出错

WEEWXX, 2024-02-17:
确实启用梯子，为啥tailscale更新才会出问题了，请问一下是什么逻辑呢，谢谢，我好检查。

CH3NGYZ, 2024-02-17:
> 确实启用梯子，为啥tailscale更新才会出问题了，请问一下是什么逻辑呢，谢谢，我好检查。

这个问题我也遇到过，我的解决办法是在安装脚本里植入停止翻墙工具的命令，等待更新完毕再启动，防止翻墙工具对路由表更改导致dns解析失败，issues里也有其他人遇到过类似的问题，他是重启了下就好了

WEEWXX, 2024-02-17:
请问加啥呢，哈哈用的oc

CH3NGYZ, 2024-02-17:
> 请问加啥呢，哈哈用的oc

https://github.com/CH3NGYZ/tailscale-openwrt/blob/bypass_clash/usr%2Fbin%2Ftailscale

WEEWXX, 2024-02-17:
好的 我试试 谢谢

WEEWXX, 2024-02-17:
> > 请问加啥呢，哈哈用的oc
> 
> https://github.com/CH3NGYZ/tailscale-openwrt/blob/bypass_clash/usr%2Fbin%2Ftailscale
> 
> 这个是老脚本，我现在用的加代理的chinese_mainland分支的安装脚本，配合oc没问题，版本更新重启也没问题，或许和系统也有原因。
> 
> 同时尽量将tailscale启动顺序提前，最好能在clash之前启动

大佬用的什么版本能分享不呢

CH3NGYZ, 2024-02-19:
> > > 请问加啥呢，哈哈用的oc
> > 
> > https://github.com/CH3NGYZ/tailscale-openwrt/blob/bypass_clash/usr%2Fbin%2Ftailscale
> > 
> > 这个是老脚本，我现在用的加代理的chinese_mainland分支的安装脚本，配合oc没问题，版本更新重启也没问题，或许和系统也有原因。
> > 
> > 同时尽量将tailscale启动顺序提前，最好能在clash之前启动
> 
> 大佬用的什么版本能分享不呢

就是本仓库另一个分支的脚本，里面有停止clash和启动clash的命令，你自己看看

CH3NGYZ, 2024-03-05:
原因:  https://github.com/CH3NGYZ/tailscale-openwrt/issues/11#issuecomment-1974802872  https://github.com/CH3NGYZ/tailscale-openwrt/issues/11#issuecomment-1978071981

已修复

closed 2024-03-05 by CH3NGYZ