CH3NGYZ/tailscale-openwrt #11: 作者你好，同样遇到dns问题

Always67, 2024-02-29:
系统是istoreos22.03.6，pw是1.47，使用mainland分支脚本。
遇到问题是软路由重启后，dns就出错。
尝试设置pw延迟3分钟启动问题依旧，现在卸载了pw，还是不行。
上一次遇到这问题是靠刷机解决的，不知道增加路由表这个操作是在哪里进行的？

root@iStoreOS:~# nslookup baidu.com
;; connection timed out; no servers could be reached

CH3NGYZ, 2024-02-29:
感觉像是tailscale和翻墙工具争抢DNS，因为tailscale也是需要用到DNS相关的服务

CH3NGYZ, 2024-02-29:
请先对翻墙工具设置不要拦截dns请求，排除是不是翻墙工具拦截DNS导致的问题，如果还是未解决，那就应该考虑路由表的问题了

WEEWXX, 2024-02-29:
作者好，我就是上个求职者。我看了是tailsacle修改了istoreos的  /etc/resolv.conf这个配置文件，楼上的如果dns错误先把配置文件修改成
search lan
nameserver 127.0.0.1
nameserver ::1
不会的话就登录了执行
echo "search lan" > /etc/resolv.conf
echo "nameserver 127.0.0.1" >> /etc/resolv.conf
echo "nameserver ::1" >> /etc/resolv.conf
然后重启，DNS就恢复正常了。
但是很奇怪，如果用istoreos的市场里的tailsacle安装也会修改这个文件，但是路由器又能正常工作，我看了作者的配置文件，又没有什么特别的配置，想不明白了。

WEEWXX, 2024-02-29:
还有想起一件事，我有多个软路由，在测试的时候上下级路由都安装了作者的版本，都启动了oc，神奇的事情发生了，下级DNS正常，上级不正常。

WEEWXX, 2024-02-29:
> 还有想起一件事，我有多个软路由，在测试的时候上下级路由都安装了作者的版本，都启动了oc，神奇的事情发生了，下级DNS正常，上级不正常。

我分析启动顺序应该是先OC，再tailsacle。有空试试

Always67, 2024-03-01:
/etc/resolv.conf 这个文件被改成了

nameserver 100.100.100.100
search taila36a8.ts.net lan

盲猜是每次重启都会重新下载tailscale，但是因为此时tailscale并没有运行，所以出错？不是很懂这一块

Always67, 2024-03-01:
再次复现，即使没有安装代理软件，同样会遇到dns错误，不知道可以提供哪些信息供作者分析。

CH3NGYZ, 2024-03-01:
> /etc/resolv.conf 这个文件被改成了
> 
> nameserver 100.100.100.100
> search taila36a8.ts.net lan
> 
> 盲猜是每次重启都会重新下载tailscale，但是因为此时tailscale并没有运行，所以出错？不是很懂这一块

请提供一下img文件，我安装复现一下，看看怎么回事

Always67, 2024-03-01:
iStoreOS 22.03.6 2024022310

https://fw10.koolcenter.com:5000/iStoreOS/x86_64_efi/istoreos-22.03.6-2024022310-x86-64-squashfs-combined-efi.img.gz

WEEWXX, 2024-03-01:
https://www.istoreos.com/
这是官网，今天固件有更新

CH3NGYZ, 2024-03-01:
> https://www.istoreos.com/ 这是官网，今天固件有更新

> iStoreOS 22.03.6 2024022310
> 
> https://fw10.koolcenter.com:5000/iStoreOS/x86_64_efi/istoreos-22.03.6-2024022310-x86-64-squashfs-combined-efi.img.gz

我用的是 EzOpWrt Ipv6-Super-R2023.10.01 by Sirpdboy 和 LEDE的 OpenWrt R23.4.1 / LuCI Master (git-23.087.57160-8186ccc)， 正常使用都没有问题，包括重启也不会丢失连接， 感觉应该是这个系统，有一些其他设置，导致tailscale可执行文件启动时与openclash/passwall等需要DNS拦截的软件争抢修改`/etc/resolv.conf`的缘故，试试吧clash的启动优先级设定为99，tailscale为90，看看问题是否依旧

CH3NGYZ, 2024-03-01:
> https://www.istoreos.com/ 这是官网，今天固件有更新

iStoreOS 22.03.6 2024030112
这是我今天测试的情况， 重启是没问题的，tailscale和openclash都没有出现问题， 此版本我先保留， 等待下一次tailscale升级的时候再看看会不会启动不了
但是按照截图中所示，tailscale确实是对系统做了一些操作。
[image]

CH3NGYZ, 2024-03-01:
> iStoreOS 22.03.6 2024022310
> 
> https://fw10.koolcenter.com:5000/iStoreOS/x86_64_efi/istoreos-22.03.6-2024022310-x86-64-squashfs-combined-efi.img.gz

> https://www.istoreos.com/ 这是官网，今天固件有更新

iStoreOS 22.03.6 2024030112
这是我今天测试的情况， 重启是没问题的，tailscale和openclash都没有出现问题， 此版本我先保留， 等待下一次tailscale升级的时候再看看会不会启动不了
但是按照截图中所示，tailscale确实是对系统做了一些操作。
[image]

Always67, 2024-03-01:
难道是tailscale和passwall有冲突吗？另外可能因为我还安装了smartdns，我试试纯净安装还会不会出问题。而且我其中一个软路由下的网心云从公网型变成端口限制了，不知道是不是有一定联系

Always67, 2024-03-02:
@CH3NGYZ @WEEWXX 我想我找到问题所在了，我在tailscale控制中心启用了magicdns，这就导致tailscale会改写系统的/etc/resove.conf，由于此时tailscale并没有启动，所以100.100.100.100这个dns无效。系统进入死循环了。解决办法一是在tailscale控制台禁用magicdns，二是使用tailscale up [...] --accept-dns=false参数。还有一个不美丽的办法，就是用脚本删除tailscale创建的/etc/resove.conf文件，重启网络。这时候tailscale可以正常下载，并且覆写/etc/resove.conf。

Always67, 2024-03-02:
@WEEWXX istoreos的tailscale版本较低，大概不支持magicdns，因此不受影响。作者也许在tailscale的控制台没有启用magicdns，因此也不受影响。我在禁用magicdns之后，可以正常使用了。

CH3NGYZ, 2024-03-02:
> @WEEWXX istoreos的tailscale版本较低，大概不支持magicdns，因此不受影响。作者也许在tailscale的控制台没有启用magicdns，因此也不受影响。我在禁用magicdns之后，可以正常使用了。

是的，我没有使用magicdns

Always67, 2024-03-02:
> > @CH3NGYZ @WEEWXX 我想我找到问题所在了，我在tailscale控制中心启用了magicdns，这就导致tailscale会改写系统的/etc/resove.conf，由于此时tailscale并没有启动，所以100.100.100.100这个dns无效。系统进入死循环了。解决办法一是在tailscale控制台禁用magicdns，二是使用tailscale up [...] --accept-dns=false参数。还有一个不美丽的办法，就是用脚本删除tailscale创建的/etc/resove.conf文件，重启网络。这时候tailscale可以正常下载，并且覆写/etc/resove.conf。
> 
> 那要不这样吧，我在下载器的那个脚本里面内置一下 echo 223.5.5.5 到脚本里面

感谢，期待脚本更新后测试

CH3NGYZ, 2024-03-02:
> 感谢，期待脚本更新后测试

试试，我已经更新最新脚本，看起来应该没有问题，但是因为我现在身边只有一只手机，所以没有办法通过路由器测试，麻烦你测试好了，如果没问题可以反馈一下，而且cdn也有一定的延时，可以等十分钟至半小时之后再试

Always67, 2024-03-03:
> 试试，我已经更新最新脚本，看起来应该没有问题，但是因为我现在身边只有一只手机，所以没有办法通过路由器测试，麻烦你测试好了，如果没问题可以反馈一下，而且cdn也有一定的延时，可以等十分钟至半小时之后再试

测试过了，打开magicdns也可以正常使用了。但是NAT类型的问题似乎无解，尝试了添加tailscale接口以及防火墙，但是软路由本身通过docker安装的网心云还是显示对称型。但是软路由下的电脑，使用软件测试却是全锥型

WEEWXX, 2024-03-04:
有个疑问  istroeos的市场提供的tailsacle为1.3.2版本，确定没有启用magicdns（需要1.4.0及以上才支持）。但是依然会修改 /etc/resolv.conf这个配置，100.100.100.100还是被写入了，所以问题出在哪儿呢

CH3NGYZ, 2024-03-05:
> 有个疑问 istroeos的市场提供的tailsacle为1.3.2版本，确定没有启用magicdns（需要1.4.0及以上才支持）。但是依然会修改 /etc/resolv.conf这个配置，100.100.100.100还是被写入了，所以问题出在哪儿呢

我推测是因为开机后tailscale_downloader安装tailscale完毕之后, 修改了系统DNS列表指向tailscale,这样本没有问题, 但是由于本仓库的脚本是系统 ***每次*** 启动时都会自动下载最新版本的tailscale, 这时候DNS还是第一次启动后修改的100.100.100.100, 又因为此时tailscale并没有启动所以才导致下载脚本无法访问网络, 因此我在下载器内添加了重置DNS命令之后脚本才能正常访问网络. 而istoreos市场安装的是固化在系统内的, 重启不会丢失, 所以就算/etc/resolv.conf被更改了也能启动链接tailscale后台 (猜测最有可能的原因就是这个)

WEEWXX, 2024-03-05:
> > 有个疑问 istroeos的市场提供的tailsacle为1.3.2版本，确定没有启用magicdns（需要1.4.0及以上才支持）。但是依然会修改 /etc/resolv.conf这个配置，100.100.100.100还是被写入了，所以问题出在哪儿呢
> 
> 我推测是因为tailscale_downloader第一次安装完tailscale之后, tailscale修改了系统DNS列表指向自身, 但是由于本仓库的脚本是系统 ***每次*** 启动时都会自动下载最新版本的tailscale, 这时候DNS还是第一次启动后修改的100.100.100.100, 又因为本脚本安装的tailscale是在/tmp目录内, 重启丢失, 所以此时tailscale并没有启动, 才导致下载脚本无法访问网络, 因此我在下载器内添加了重置DNS命令之后tailscale_downloader脚本才能正常访问网络. 而istoreos市场安装的是固化在系统内的, 重启不会丢失, 所以就算/etc/resolv.conf被更改了也能启动链接tailscale后台 (猜测最有可能的原因就是这个)

明白了，这个脚本也完善了，感谢作者。

CH3NGYZ, 2024-03-05:
> 明白了，这个脚本也完善了，感谢作者。

我现在也明白为什么当时我开了clash下载器才好使了, 因为clash开了"路由器自身代理"会修改dns指向127.0.0.1, 这样脚本才能进行域名解析, 否则依旧是之前的100.100.100.100 

> 测试过了，打开magicdns也可以正常使用了。但是NAT类型的问题似乎无解，尝试了添加tailscale接口以及防火墙，但是软路由本身通过docker安装的网心云还是显示对称型。但是软路由下的电脑，使用软件测试却是全锥型

网心云的Docekr容器网络类型是Host吗, 如果不是, 改成Host试试

CH3NGYZ, 2024-03-05:
已修复 : https://github.com/CH3NGYZ/tailscale-openwrt/commit/c8ed0d07c71ab32450b2fed422d6dcab5766f445

closed 2024-03-05 by CH3NGYZ

Always67, 2024-03-15:
> 我现在也明白为什么当时我开了clash下载器才好使了, 因为clash开了"路由器自身代理"会修改dns指向127.0.0.1, 这样脚本才能进行域名解析, 否则依旧是之前的100.100.100.100
>
>
> 网心云的Docekr容器网络类型是Host吗, 如果不是, 改成Host试试

是host，不过这好像跟脚本无关。使用zerotier和tailscale组网都会遇到这个问题。可能是网心云检测的问题