CH3NGYZ/tailscale-openwrt #13: 今天更新后 无法跨子网访问了

gzxiexl, 2024-05-09:
我的应用场景是家里和办公室openwrt下安装tailscale。
分别通告了子网
tailscale up --accept-routes=true --accept-dns=false --advertise-exit-node --advertise-routes=192.168.6.0/24  --reset
目前两台路由进ssh可以访问对方，但是局域网内设备无法访问另外一个子网下的设备了，原来是没问题的。

gzxiexl, 2024-05-09:
路由A 10.0.0.1/24
配置参数：tailscale up --accept-routes=true --accept-dns=false --advertise-exit-node --advertise-routes=10.0.0.0/24  --reset
路由B：192.168.6.1/24
启动参数：tailscale up --accept-routes=true --accept-dns=false --advertise-exit-node --advertise-routes=192.168.6.0/24  --reset

现在路由B ssh ping得通10.0.0.1 10.0.0.254（NAS）
但是B下的局域网设备不能连10.0.0.254了 似乎就是路由不会转发了 不知道是不是这么理解

gzxiexl, 2024-05-09:
爬了下官方wiki 
By default, only outbound connections from the tailnet to the subnet are allowed. To initiate connections from the subnet into the tailnet, you will also need to set --stateful-filtering=false
默认情况下，仅允许从尾网到子网的出站连接。要启动从子网到尾网的连接，您还需要设置 --stateful-filtering=false .

启动参数加上 --stateful-filtering=false 就正常了

closed 2024-05-13 by gzxiexl

CH3NGYZ, 2024-05-15:
sorry最近没有看仓库, 关于这个问题你可以fork本仓库, 然后修改对应的脚本, 由于大部分用户并不需要修改这一部分, 只能是你自己定制一下脚本了

CH3NGYZ, 2024-05-15:
> 爬了下官方wiki By default, only outbound connections from the tailnet to the subnet are allowed. To initiate connections from the subnet into the tailnet, you will also need to set [...]

sorry 最近没有看仓库, 忽略了你的issue, 关于这个问题, 其实本仓库只是下载安装并以默认参数启动, 至于这些具体的功能你可以直接在ssh里用你说的参数启动就可以设置成功了, 因为配置文件是写在/etc/tailscale里的, 更新tmp里的主程序不会删除配置文件.

gzxiexl, 2024-05-15:
我注意到官方已经支持了 autoupdate 但是貌似没有效果

CH3NGYZ, 2024-05-16:
因为本仓库的脚本是每次启动时从本仓库下载最新的可执行文件, 所以官方的更新也就对固化安装在系统内的tailscale起作用, 本仓库不用他那个自动更新

CH3NGYZ, 2024-05-16:
> 我注意到官方已经支持了 autoupdate 但是貌似没有效果

因为本仓库的脚本是每次启动时从本仓库下载最新的可执行文件, 所以官方的更新也就对固化安装在系统内的tailscale起作用, 本仓库不用他那个自动更新