CH3NGYZ/tailscale-openwrt #14: 提个小建议，可以修改一下脚本可以少装个timeout的包

waltermanpro, 2024-07-05:
if wget -T $timeout_seconds -q $proxy_zip_url -O - | tar xzf - -C / >/dev/null 2>&1; then

get_version=$(wget -T $version_timeout -qO- "$version_proxy_url" | grep 已从上游同步 | head -1 | cut -d'%' -f 2)

if wget -T $version_timeout -c -qO- $zip_proxy_url | tar xzvf - -C /tmp -T /tmp/tailscale_${version}_files.txt > /dev/null 2>&1; then

我使用chatgpt修改的，没安装timeout的包，可以在armv7l的docker中的openwrt正常运行

waltermanpro, 2024-07-05:
建议在tailscale_downloader中添加echo "nameserver 100.100.100.100" >> /etc/resolv.conf，以保证tailscale dns能正确运行(我docker里运行的openwrt需要，其他版本不清楚)

waltermanpro, 2024-07-05:
还有tailscale需要的那三个依赖好像，新版openwrt都有，ImmortalWrt 18.06-k5.4-SNAPSHOT r12339-8b50c1df21 / LuCI openwrt-18.06-k5.4 branch (git-22.312.35760-7afcfb2)，我用的这么老的都有。

CH3NGYZ, 2024-07-05:
好的, 安装那三个包也是怕有些系统没有. 我接下来会先检测系统有没有再安装. 至于`echo "nameserver 100.100.100.100" >> /etc/resolv.conf` 在另一个issue里有提到过 https://github.com/CH3NGYZ/tailscale-openwrt/issues/11#issuecomment-1974802872 待会我会一并添加

closed 2024-07-05 by CH3NGYZ