CH3NGYZ/tailscale-openwrt #18: 请问有办法让其安装到闪存而不是内存吗？

lingzs13, 2024-09-08:
我的路由器wr30u闪存还有50m，但内存只有36m如果把文件放到内存就无法正常启动。

CH3NGYZ, 2024-09-08:
那最好你手动安装, 脚本下载zip还要解压, 内存闪存都不够

CH3NGYZ, 2024-09-08:
在release下载对应架构和版本的zip文件, 不管用什么方法, 将 tailscale 和 tailscaled 放入 /usr/bin 内, 设定可执行权限, 在将本仓库 /usr/init.d/tailscale 复制到路由器一样的路径内, 这样就是固化的, 而不是本项目每次开机都会下载最新版本.

CH3NGYZ, 2024-09-08:
但即使是手动安装, arm64的那两个文件也都47.8MB了, 应该塞不下的吧

closed 2024-09-08 by CH3NGYZ

lingzs13, 2024-09-08:
> 但即使是手动安装, arm64的那两个文件也都47.8MB了, 应该塞不下的吧
感谢🙏大佬的回复，根据你的讲究我已经配置完成了，目前看来能够正常使用，至于磁盘装不下的问题我的也有点看不懂这个系统的检测机制了💦他的这两个文件加起来有`47.8MB`但在web端显示加上我之前用的才到了`43.18MIB`。
我用的还是ImmortalWrt的官方固件

CH3NGYZ, 2024-09-08:
能用就行

CH3NGYZ, 2024-09-08:
> > 但即使是手动安装, arm64的那两个文件也都47.8MB了, 应该塞不下的吧
> 
> 感谢🙏大佬的回复，根据你的讲究我已经配置完成了，目前看来能够正常使用，至于磁盘装不下的问题我的也有点看不懂这个系统的检测机制了💦他的这两个文件加起来有`47.8MB`但在web端显示加上我之前用的才到了`43.18MIB`。 我用的还是ImmortalWrt的官方固件

能用就行啦, 就是你得注意剩余空间要不然不够了, 影响系统启动

viv1dst, 2024-11-19:
> 那最好你手动安装, 脚本下载zip还要解压, 内存闪存都不够 手动安装固化版本:
> 
> 1.在 [releases](https://github.com/CH3NGYZ/tailscale-openwrt/releases) 下载对应架构和版本的zip文件并解压到电脑 2.不管用什么方法, 将 `tailscale` 和 `tailscaled` 复制到路由器 `/usr/bin/` 路径内 3.增加两个文件的可执行权限 `chmod +x /usr/bin/tailscale` 和 `chmod +x /usr/bin/tailscaled` **4.将本仓库 `/usr/init.d/tailscale` 文件复制到路由器 `/usr/init.d/**` 路径内 5.`/etc/init.d/tailscale enable` 6.`/etc/init.d/tailscale start` 稍等10秒 7.`tailscale up`
> 
> 即可. 这样就是固化的, 而不是本项目每次开机都会下载最新版本.

请问第4步是不是写错了 应该是/etc/init.d

CH3NGYZ, 2024-11-20:
> 请问第4步是不是写错了 应该是/etc/init.d

-.- 还真是, 谢谢你, 我改回来

CH3NGYZ, 2025-04-16:
> 能否安装脚本提供选项 安装到内存或者是硬盘里🤔 一键脚本还是过瘾

我能否采访一下，你的op设备是什么架构，内存以及存储余量还有多少，因为我还在想如果有了UPX压缩之后，只有5MB大小的体积要不要只保留本地安装算了