CH3NGYZ/tailscale-openwrt #19: 下载失败，梯子已经开了

rotent, 2024-09-10:
[image]
[image]

CH3NGYZ, 2024-09-11:
检查DNS

CH3NGYZ, 2024-09-11:
此问题非脚本问题, 是你路由器自身网络问题, 在路由器内 `nslookup ghproxy.cc` 检查一下

rotent, 2024-09-11:
因为我学校是校园网，禁止了公共dns
[image]

rotent, 2024-09-11:
主要是前面的流程都没问题，到了下载tailscale本体的时候就不行了
[image]

CH3NGYZ, 2024-09-11:
试试路由器内 用curl或wget下载这个文件 , 看看多长时间下载完, 脚本内置5秒, 5秒内下不完就视为失败.
`wget -O /dev/null https://ghproxy.cc/https://github.com/CH3NGYZ/tailscale-openwrt/releases/download/1.72.1/tailscale_1.72.1_arm64.tgz
`

`curl -o /dev/null https://ghproxy.cc/https://github.com/CH3NGYZ/tailscale-openwrt/releases/download/1.72.1/tailscale_1.72.1_arm64.tgz
`

CH3NGYZ, 2024-09-11:
我已经修改超时为30秒, 请再试一次

rotent, 2024-09-11:
成功了！感谢！！