CH3NGYZ/tailscale-openwrt #22: 竞斗云一键失败

laozhudu, 2024-10-13:
设备信息<html>
<body>

主机名 | OpenWrt
-- | --
型号 | P&W R619AC 128M
CPU信息 | ARMv7 Processor rev 5 (v7l) x 4 (716MHz, 51.0°C 55.0°C )
目标平台 | ipq40xx/generic
固件版本 | OpenWrt 21.02-SNAPSHOT / LuCI 21.02 Lienol git-c2aa27f
内核版本 | 5.4.238

</body>
</html>

root@OpenWrt:~# uname -m
armv7l
root@OpenWrt:~# wget -O- https://ghproxy.cc/https://raw.githubusercontent.com/CH
3NGYZ/tailscale-openwrt/chinese_mainland/install.sh | sh
--2024-10-13 17:49:03--  https://ghproxy.cc/https://raw.githubusercontent.com/CH        3NGYZ/tailscale-openwrt/chinese_mainland/install.sh
Resolving ghproxy.cc... 104.20.21.207
Connecting to ghproxy.cc|104.20.21.207|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/plain]
Saving to: 'STDOUT'

-                       [ <=>                ]   4.44K  --.-KB/s    in 0.001s

2024-10-13 17:49:04 (4.20 MB/s) - written to stdout [4549]

Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/ipq40x        x/generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/ipq40x        x/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_c        ortex-a7_neon-vfpv4/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_c        ortex-a7_neon-vfpv4/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/kenzo/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/kenzo/Packages.gz

Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/small/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/small/Packages.gz

Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/kenzo/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/small/Packages.gz, wget returned 8.
INSTALL: 包 libustream-openssl 未安装，开始安装...
Installing libustream-openssl20201210 (2022-01-16-868fd881-2) to root...
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/base/libustream-openssl20201210_2022-01-16-868fd881-2_arm_cortex-a7_neon-vfpv4.ipk
Collected errors:
 * check_data_file_clashes: Package libustream-openssl20201210 wants to install file /lib/libustream-ssl.so
        But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package libustream-openssl.
INSTALL: 安装 libustream-openssl 失败，请手动安装或检查原因

laozhudu, 2024-10-13:
麻烦看看谢谢

CH3NGYZ, 2024-10-13:
`check_data_file_clashes: Package libustream-openssl20201210 wants to install file /lib/libustream-ssl.so
But that file is already provided by package * libustream-wolfssl20201210`

安装`libustream-openssl`时某些文件已存在于`libustream-wolfssl20201210`包内, 文件冲突无法安装, 请自行修改脚本跳过安装`libustream-openssl`

CH3NGYZ, 2024-10-15:
修改了安装脚本，现在脚本可以直接跳过安装失败的包，麻烦重新下载运行一下并告知我是否成功

closed 2024-10-15 by CH3NGYZ

laozhudu, 2024-10-15:
INSTALL: 包 libustream-openssl 未安装，开始安装...
Installing libustream-openssl20201210 (2022-01-16-868fd881-2) to root...
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/base/libustream-openssl20201210_2022-01-16-868fd881-2_arm_cortex-a7_neon-vfpv4.ipk
Collected errors:
 * check_data_file_clashes: Package libustream-openssl20201210 wants to install file /lib/libustream-ssl.so
        But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package libustream-openssl.
INSTALL: 安装 libustream-openssl 失败，跳过该包，如果无法正常运行 tailscale，请排查是否需要手动安装该包
INSTALL: 包 ca-bundle 已安装，跳过
INSTALL: 包 kmod-tun 已安装，跳过
INSTALL: 包 coreutils-timeout 未安装，开始安装...
Installing coreutils-timeout (8.32-6) to root...
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/packages/coreutils-timeout_8.32-6_arm_cortex-a7_neon-vfpv4.ipk
Configuring coreutils-timeout.
INSTALL: 包 coreutils-timeout 安装成功
INSTALL: ------------------
下载安装脚本失败，尝试下一个代理
---------------------------
INSTALL: ------
下载安装脚本成功!
---------------
INSTALL: --------------
正在启动 Tailscale 下载器
-----------------------
DOWNLOAD: -----------------------------------
通过 https://mirror.ghproxy.com/https://github.com 获取版本号：1.76.0
---------------------------------------------
DOWNLOAD: ---------------------------------
通过 https://mirror.ghproxy.com/https://github.com 下载并解压成功!
-------------------------------------------
INSTALL: ----------------
正在启动 Tailscale 后台服务
-------------------------
INSTALL: ----------------
正在启动 Tailscale 前台程序
成功了，谢谢