CH3NGYZ/tailscale-openwrt #29: 发现异常

qingyuan0o0, 2025-02-28:
root@CR8808-OP:~# tailscale ping 750e
pong from 750e (100.64.0.3) via DERP(cn) in 34ms
pong from 750e (100.64.0.3) via DERP(cn) in 23ms
pong from 750e (100.64.0.3) via DERP(cn) in 24ms
pong from 750e (100.64.0.3) via DERP(cn) in 21ms
pong from 750e (100.64.0.3) via DERP(cn) in 20ms
pong from 750e (100.64.0.3) via DERP(cn) in 23ms
error Post "http://local-tailscaled.sock/localapi/v0/ping?ip=100.64.0.3&size=0&type=disco": read unix @->/var/run/tailscale/tailscaled.sock: read: connection reset by peer:
root@CR8808-OP:~# tailscale ping 750e
failed to connect to local tailscaled; it doesn't appear to be running
root@CR8808-OP:~# uname -m
armv7l

测试版本是1.80.2,主机是小米cr8808,openwrt,
异常是会自动结束进程

CH3NGYZ, 2025-03-04:
看起来是后台服务没有启动或者出错，看看日志能看到更多报错吗

qingyuan0o0, 2025-03-05:
> 看起来是tailscaled后台服务没有启动或者出错，看看日志能看到更多报错吗

我单独运行tailscaled --verbose=1，等待几秒后，看到一行输出是kill开头的，被结束进程了，因为我已经卸载1.80.2，现在opkg安装的是1.24的版本

CH3NGYZ, 2025-04-12:
> > 看起来是tailscaled后台服务没有启动或者出错，看看日志能看到更多报错吗
> 
> 我单独运行tailscaled --verbose=1，等待几秒后，看到一行输出是kill开头的，被结束进程了，因为我已经卸载1.80.2，现在opkg安装的是1.24的版本

等等我的新脚本吧，那个新脚本可以把可执行文件体积压缩到很小，到时候会更可靠一些

CH3NGYZ, 2025-04-19:
代码初步编写完毕, 请移步[此仓库](https://github.com/CH3NGYZ/small-tailscale-openwrt)

closed 2025-04-19 by CH3NGYZ