CH3NGYZ/tailscale-openwrt #31: glinet-tailscale-updater确实牛 缩小了好多 请大佬赶紧合并吧

gzxiexl, 2025-04-16:
我fork了glinet-tailscale-updater后尝试改了下 就编译了amd64和arm64两种架构
这是我fork后的仓库
https://github.com/gzxiexl/glinet-tailscale-updater

[image]

好小的说 太牛了

CH3NGYZ, 2025-04-16:
是的，正在编写代码咯  [新测试仓库](https://github.com/CH3NGYZ/ts-test/releases)
[image]

CH3NGYZ, 2025-04-16:
其实光弄release还不够，我还要同步修改安装脚本，还得根据国内情况进行优化，比如说想关闭更新，指定某个版本，可选安装到内存或闪存等等，关于本地化你还有什么点子吗？

gzxiexl, 2025-04-16:
你知不知道有个ShellCrash的项目 https://github.com/juewuy/ShellCrash
我建议可以参考下这个项目关于mihomo 内核升级、降级的思路，基本上各种情况都考虑到了，甚至本地上传到临时文件夹也可以更新。
本地化主要考虑一是代理加速，我公司有个100M上下行的电信固定ip宽带，我放了一台nas常年不关机，不知道有没有意义，如果有需要可以提供下分流；
二是有没有可能脚本可以协助配置tailscale up后的一些参数，降低tailscale的使用成本。例如我的配置参数是:
`tailscale up --stateful-filtering=false --accept-routes=true --accept-dns=false --advertise-exit-node --advertise-routes=192.168.6.0/24  --reset`
公司和家里相互开启子网互访，不确定这个参数有没有问题或者其他的优化的地方，就我自己用下来没什么大问题。

CH3NGYZ, 2025-04-16:
> 你知不知道有个ShellCrash的项目 https://github.com/juewuy/ShellCrash 我建议可以参考下这个项目关于mihomo 内核升级、降级的思路，基本上各种情况都考虑到了，甚至本地上传到临时文件夹也可以更新。 本地化主要考虑一是代理加速，我公司有个100M上下行的电信固定ip宽带，我放了一台nas常年不关机，不知道有没有意义，如果有需要可以提供下分流； [...]

会参考这个脚本，我感觉他这个下载函数写的挺不错的。还有鄧简单的一键完成脚本写完能正常使用后再考虑做一个脚本用来一键通过一些配置项生成tailscale启动命令。

closed 2025-04-19 by CH3NGYZ